armageddon421/electrodacus-esp32 #22: In Client mode, electrodacus-esp32 needs to respond to ARP Requests

mark6sf, 2020-10-16:
In client mode, when connected to a WiFi access point, the electrodacus-esp32 (e-esp32)does not respond to ARP Request messages.  Instead it sends out a Gratuitous ARP once a minute to provide its MAC address information. This maybe fine when the e-esp32 acts as a AP but not when it is an AP client. 

Not responding to ARP Requests has 2 negative consequences:
1. When trying to connect a browser, if the e-esp32's MAC address is not in the caller's ARP cache, the caller has to wait for the e-esp32 to send a Gratuitous ARP message.  This delays the start of the connection and often causes the browser to timeout with a "host unreachable" error.  This is the case for tested Windows 10 and MACs machines.

2.  Tested Android based phones do not appear to process the Gratuitous ARP messages at all and are therefore not able to connect. 

Solution: The e-esp32 needs to respond to an ARP Request by sending an ARP Reply. Once this is done the Gratuitous ARP message is unnecessary in client mode. 

Attached is a zipped WireShark capture from a MAC showing no replies to ARP requests but connecting when a Gratuitous ARP message was sent by the e-esp32.

Please let me know if you need further information.

Mark

[201016.0453 Waiting for Gratitous ARP.zip](https://github.com/armageddon421/electrodacus-esp32/files/5393626/201016.0453.Waiting.for.Gratitous.ARP.zip)

armageddon421, 2020-10-16:
Hi Mark,

I'm sorry to hear you are having trouble. Thank you for your investigation.

A quick search showed that this is a general Problem with the wifi stack on the esp32. It has either to do with modem sleep modes for power saving, which I have not consciously activated, or is something that can not easily be solved.

Usually, the router or network infrastructure will cache the gratuitous ARP the esp32 sends after initially connecting and there will be no problems with that. To mitigate any problems that may arise nonetheless, the re-sending every minute was established. This is generally considered to be an "ugly" workaround.

I have personally not had any problems regarding this and could always connect to the various esp32 and esp8266 devices I own from any windows/linus/android machine immediately.

Please let me know if this actually renders the device unusable for you, if it also happened with a different firmware version (if applicable) and what router you are using. I will try to investigate if for some reason the modem sleep mode is activated in the firmware by default. Hopefully, I can somehow recreate the problem.

Please note that I only work on this project intermittently, so it might be a short while until I can come up with a solution that may work for you.

Greetings,
Robert

mark6sf, 2020-10-16:
Robert,

First let me say that this is not causing any real problem for me since I am home and can access the SBMS0 with a PC or MAC. Also, no time issue.  I think it could be more of a problem for those doing mobile applications where they are more likely to use cell phones if those phones don't handle graduations ARPs 

Thanks for pointing out this is a known issue with the esp32 and maybe related to sleep modes.  I will do more research on that.

I should note in reproducing the problem I would issue an "arp -a -d" to flush the devices ARP cache, except for the android phone. The phone is Android version 9.

I have 4 separate WiFi APs with different SSIDs from several different companies but the all share the same IP address space 192.168.44/24.  The SBMS0 is connected to a TP-Link  EAP-225 on SSID mokg.  They are all set as APs and do no routing.  The router is a separate Untangle NG Firewall.  The Untangle does DHCP but does not seem to be involved in ARP caching, at least for devices on the same subnet.  I did notice one AP at 192.168.44.10  (not the one with the SBMS0) responds to the gratuitous ARP by sending out an ARP request but of course gets no response from the esp32.

The results were the same regardless of which WiFI AP the calling device was using including mokg.  So apparently in my network there are no helpers to cache and respond  to ARP requests.  If it is the case that WiFI  APs that are also routers do this function it would explain why others haven't seen this issue and make it less of a problem. 

I mainly wanted you to be aware of the issue.  I will try to do more testing and let you know any results.

Thanks,
Mark

kiss985632, 2020-10-18:
> In client mode, when connected to a WiFi access point, the electrodacus-esp32 (e-esp32)does not respond to ARP Request messages. Instead it sends out a Gratuitous ARP once a minute to provide its [...]

mark6sf, 2020-11-18:
Hi Robert,

As a follow up I have done additional testing on the ARP-response issue.

I confirmed that when the SBMS0 was on the main network the main router cached its ARP address.  However, when the querying device is on the same subnet the router assumes the target device will respond to the ARP-request,

I then set up a separate WiFi router just for SBMS0 with a separate address space.  As expected this fixed the ARP-response problem for Android phones as the router handled the routing between the subnets.

In researching the ESP32, it appears the WiFi modem is turned off most of the time in all but “active” mode and that the default setting for the ESP32 is a sleep mode.  I found some references that indicate that the ESP32 will respond to ARP-requests in “active” mode. The following link describes the functions and power requirements of various modes.

https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/

While the power requirements for active mode are high, they are still certainly less than having to run a separate WIFI router.  I recommend that an option be added to the WiFi configuration web page to select between “active” and one of the low power sleep modes with the default being the low power mode.

armageddon421, 2020-11-18:
Thank you for all that information and investigation. from my understanding I have to actively activate the sleep modes of the wifi unit, which I haven't done yet. So I imagine the device is in "active" mode normally right now.

I have no troubles accessing it with linux devices and my android phone right now so I cannot reproduce your issue.

If you could detail your setup I can possibly try to recreate it with the spare sbms and spare wifi router hardware I have.

mark6sf, 2021-01-02:
In recent months the problem of access from my Android phone has been intermittent, sometimes it works and sometimes it doesn't.  It has been difficult to get packet traces but I feel it may be due to one particular TP-Link access point that does not consistently forward gratuitous ARP messages.  Since no one else has reported this problem I am going to assume it is something unique to my network and that we should ignore it for now.  Thanks for your help. I have an enhancement request I will submit separately.

closed 2021-01-10 by mark6sf