armageddon421/electrodacus-esp32 #23: Increase sturdiness of raw data parsing/storage

armageddon421, 2020-11-10:
- Add list of valid variable names and check after parsing
- check for invalid symbols in variable names during parsing

```
var  sbms="#%0L6>N$#&*";
var ##="###*]i###.BI###############.BI###(1`###*K<";
var Btn="####'/))&#####$}###72$A(###@Z###1)%&$'%+####*%############%)%$#$$#(##############%&%$$$0$############%$#####-)%#########[DNl3552.../...-..-(---...-....(#,/,(..*-$##.#(,###(##.*#,..[SWSW[SWS[___W[S_[_WW[S[d_[dW_WSW[S_WW[_[[[[SWS[___SW[_[h[[[";
var Btp="67'2############uZ(#####P@'##oN3##########&)##%)*/00.$#############&,,,,+*++#############+.,..,(######################%#################################+#########K2#,##/}_#ZC##(###############################################################";
var ELd="###$*.((&######|'#&61#@*###@Y8%#0($&$&$+&###/$############%($#####'##############$%%###0#############$######,($#########eCQ|).-(###$######################$#####################################################################################";
var Ld ="####################################################################################################################################################################################'####'###'+++#'#+'+##'#'0+'0#+###'#+##'+''''###'+++##'+'4'''";
var PV1="lf-#############################################################################################################################################################################################################################################";
var PV2="################################################################################################################################################################################################################################################";
var dmppt="############################################################";
var eA="###w60##$9Xa##############$9Xa###WMm###tuh";
var eW="###.16###2:=###############2:=###*%V###-p?";
var eWbms="#;iL6>N$#&*";
var eWms="##+L6>N$#&*";
var esbms="#6oL6>N$#&*";
var gsbms="##l#######gu#'`##h#g?#&9##4#########";
var s1=['Ah','A','SBMS0  '];
var s2=[0,0,0,0,0,0,0,0,2,7,1,1];
var sbms="7.-(H;$+GbGd########GaGb*K*\\-##W#####################$A%%N(";
var xsbms="###L6>N$#&*";
```

bhey99, 2021-02-08:
what is causing real issues are invalid strings on rawdata page with character \ which are escaped into \\ 

invalid output: (one extra character)
var sbms="8%+9G\\#|GIGMGMGKGJGKGHG<(g##-#1m#######1[############&E8%N(";

valid output: 
var sbms="8%+9I1#|GGGHGHGFGFGGGIG=(i##-#6?#######6)############&CZ%N(";

armageddon421, 2021-08-31:
I think the sbms0 already escapes backslashes properly for the browser to handle them. The ESP32 code de-escapes them for the parsing for MQTT output. So actually the escape sequences are no problem right now.

commit 99e62553 (2022-08-29): add jsvar timeout and valid character check

fixed exp32 SDK to 4.4.0

closes #37
closes #23
closes #32
closed 2022-08-29 by armageddon421