armageddon421/electrodacus-esp32 #25: Does this need python2 or python3 ?

hary66, 2020-12-29:
Hi.

I tried to make the procedure from a Raspberry pi but the comm and :
`platformio run`
led to that error :+1: `Error: Could not find the package with ‘platformio/tool-ninja @ ^1.7.0’ requirements for your system ‘linux_armv7l’`

I then tried from an x86 pc but I had some error showing some python2 and python3 stuff, making me thinking that the procdedure was calling the default python interpreter which is python3 in my case when the procedure needs pyhton2 instead.

Am I right to think the procedure really needs python 2 ?
any advice to modify the default python interpreter to python2 on my pc running linux mint20 ?

armageddon421, 2020-12-29:
Hi Harold,

It seems there is something wrong with your python installation. You have a locally installed "reprlib" in your home folder for python 2.7 that your python 3 is trying to access. The tool chain should work fine with python 3, that's what the CI build is using as well.

I'm not really sure how to approach solving this as I don't know how your system is set up. Maybe try analyzing and cleaning up your python environment, remove anything that was not installed by pip or your operating system's package manager, especially things in /home/Harold/.local/lib/

Let me know if you need further assistance.

hary66, 2020-12-30:
Ok, there was something wrong with my python setup on my computer.

I finally got the ESP32 flashed and can retrieve the json data with mosquitto running on my Raspberry pi sever.

Would you mind to share your Nodered flow ?

armageddon421, 2020-12-30:
Hi Harold,

I'm glad to hear you got it working!

What are you interested in when it comes to node red?
I mostly only push most of the data to influxdb and have simple dashboard UI elements. I also use solcast to predict solar power output.

hary66, 2020-12-31:
Hi Robert and first of all, thank you very much for the esp32 update. It should give us the opportunity to have nice and easy remote control to the SBMS.

I thought you had made some kind of  Nodered dashboard. But if you didn't, that's ok, I'll sort something out myself. The only point is that it will take more time as I'm learning the thing.

I before tried to use the TICK telegraf/grafana/influxdb but never succeeded.
So it would be very nice if you would share your setup in detail and if possible even better like a tutorial.
It would be a very instructive learning case to use all these tools, solcast included.

By the way, I needed to link the esp32's Rx to FTDI USB/UART converter's Rx and esp32's Tx to FTDI USB/UART converter's  Tx to get the upload possible. Not Rx->Tx and Tx->Rx as supposed !

closed 2021-01-03 by hary66