armageddon421/electrodacus-esp32 #26: Enhancement: Show Key Status Symbols on Legacy Page

mark6sf, 2021-01-02:
I use your WiFI app extensively to monitor my power station which is two floors down in the garage and it's a great help.  Most of the time I can just leave the Legacy page on the monitor screen which shows almost everything.  However, if I want to see the status indicators such as EOC, OV, CFET, DFET, etc, I have to switch to the Home page.  It would be more convenient, especially in the case of a remotely mounted display, to have these on the Legacy page somewhere near the cell voltages.  I might suggest as a replacement for the battery outline, of course, the large SOC value is still needed. As a fixed position for each may not fit it could just show a list of the active ones.

Thanks,
Mark

bhey99, 2021-02-08:
I new here.. but i know, how it was on the local html file within the legacy sbms page. I created my own page to show CFET and DFET, excluded some values etc. 
But i do not know, how to integrate my sources here and (which is even worse) how to update the sbms0 with an updated file...
How can we work together?

soggygeek, 2021-04-22:
Marksf, the status indicators ARE on the Legacy page, just in code, right underneath the big SOC number is:
Type: 1 Cap: 540Ah Status: 20480

The Status: 20480 is the normal condition on my setup.  If it's a different number, something has changed.

You might find it interesting to set up MQTT if you have a computer you can set to sit and monitor.  I wrote up some pretty detailed notes on how to do it with node-red being the aggregator, logger, and display mechanism.  You can see here:
http://svsoggypaws.com/electrodacus.htm

Robert, thanks for all you've done!  The wifi capability (and MQTT) has made the Electrodacus infinitely more useful to us.  We are not "set it and forget it" type people, and like to peek at what's going on every day.

armageddon421, 2021-04-22:
@bhey99 you can still use a local html file if you prefer to do so. You just need to change the URL to http://<IP>/rawdata and it will work exactly the same as with the old SBMS

armageddon421, 2021-04-22:
I will not work on the legacy page. My goal is to create a new dashboard on the main page.