armageddon421/electrodacus-esp32 #27: Can't get all data from the json object with MQTT

hary66, 2021-01-03:
Hi.

Here is the json object I get from my MQTT server :
`/sbms {"time":{"year":0,"month":1,"day":3,"hour":19,"minute":5,"second":41},"soc":38,"cellsMV":[3232,3208,3228,3232,3234,3228,3228,3231],"tempInt":18.6,"tempExt":16.9,"currentMA":{"battery":-278,"pv1":820,"pv2":559,"extLoad":0},"ad2":0,"ad3":0,"ad4":0,"heat1":0,"heat2":0,"flags":{"OV":false,"OVLK":false,"UV":false,"UVLK":false,"IOT":false,"COC":false,"DOC":false,"DSC":false,"CELF":false,"OPEN":false,"LVC":false,"ECCF":false,"CFET":true,"EOC":false,"DFET":true,"delta":26}}

My question is why I can get :
msg.payload.currentMA.extLoad
but I can't get : 
msg.payload.currentMA.intLoad
which doesn't exist in the json.

Is that normal ?

`

armageddon421, 2021-01-03:
The firmware was meant for the sbms0, which does not have IntLoad. I could add it, but it'd always be zero on sbms0. I'll see if I find some time to do that later.

hary66, 2021-01-03:
Ok, I see.

That would be nice if it could be added.

By the way, that's very nice feature you have done untill now !