armageddon421/electrodacus-esp32 #28: Electrodacus SBMS0 WiFI module

fdemei, 2021-02-22:
This  afternoon I tried to configure Electrodacus WiFi . I connect my computer WiFi to SBMS APP then I settled up SBMS WiFi as client Mode. I configured SSID and Password  with home  WiFi SSID and password. I saved data then I switched off and again ON the WiFi module. But I suppose something is gone wrong. Now the WiFi symbol to SBMS display flashing , I don’t find SBMS to my home router ,the SBMS WiFi is still as access point and  IP address is still 192.168.4.1  but I’m not anymore able to open the SBMS web page.
There is some way to reset to default the WiFi Module ?

armageddon421, 2021-02-22:
Seems like your config may be damaged. Did you use any special characters in your WiFi password?
Can you still reach http://192.168.4.1/ota ?

fdemei, 2021-02-22:
Hi Armageddon,

after several attempts I was able to come back to the setting menu and I removed the Enable Wi Fi Client Check box then I resetted  WiFi module. Now work fine again, but as "WiFi AP Mode" 

The SSDI of my router is "Casa WiFi" and password is quite simple "27Novembre1954"

My router work in the same time in 2.4 Ghz mode and 5.0 Ghz mode, is the client that choose the best for him. Is it possible to use manual IP configuration instead of DHCP ?

armageddon421, 2021-02-22:
It should be able to connect to your router fine, but it will only use the 2.4ghz network. It is not possible at the moment to manually set the ip configuration, but your router should provide the option to set a fixed up address for certain devices.

If the connection to the configured network in client mode fails, the wifi module will fall back to AP mode. You should be safe to try again unless you change and forget the AP mode wifi password.

armageddon421, 2021-02-22:
It is possible that the connection to the router is somehow problematic and the wifi module is falling back to AP mode only for short periods of time. This may be why you could not properly open the configuration page. On that case you could try turning off your router or moving out of range before trying to connect directly to the sbms.

fdemei, 2021-02-23:
Hi Armageddon, 

thank you for support. I’ll live system working in AP mode .the system it will be installed on my boat where I have a special network that work with manual IP assignment. On board PC is wired to the net, then I can use WiFi to make connection with SBMS.

 

Thanks again

Best regards

 

Franco

closed 2021-03-28 by armageddon421