armageddon421/electrodacus-esp32 #30: Error: Python 3.6 or later is required for this operation.

VA2EC, 2021-03-12:
Fresh Debian 10 installation.
I follow the installation / flashing procedure and at the platformio update command i get this in red...

Error: Python 3.6 or later is required for this operation.
Please check a migration guide:
https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5

tried to follow the migration guide... without good results...

anybody cant help me on that? Thanks!

armageddon421, 2021-03-12:
Sorry but without more information I can't help you. What were the issues with that guide? What python version did your system have to begin with and what does it have now? If you have issues installing platformio, why do you create an issue here and not on platformio?

VA2EC, 2021-03-13:
Sorry about my lack of knowledge, i was just trying to get some help about this error, mainly because I was following the procedure on your page. So with some more Googleling, I found that Debian is still shipped with python2.7 as default.  
Thanks to this page: https://michlstechblog.info/blog/debian-set-python-3-as-default/

I finally compile everything (as root) and downloaded on my ESP32. Success.
(Not the original electrodacus wifi board, a ESP32 of my own)
Powered up, the blue led is blinking, a open wifi AP showed up as "SBMS". Connected, 192.168.4.1 on my browser,  no real page, just "Not found" in the upper left corner.

So if you thing my ticket was not relevant, or not useful for other, please remove it. This was the only place i found to get help. 

According to what I saw on the electrodacus forum, I think you've done a great work... but i'm stuck again.

Thanks.
Eric

armageddon421, 2021-03-14:
Hi Eric,

It seems like the spiffs is not flashed correctly so it can not read the website from it. 
Try flashing --target uploadfs again.

VA2EC, 2021-03-15:
Ok... I'm embarrassed, I didn't done this command, as soon as i saw the SBMS wifi AP appear, I was sure that I was done... Not read enough. I'm sorry, everything works as intended. Thanks again!

closed 2021-03-15 by VA2EC