armageddon421/electrodacus-esp32 #32: Legacy Page Stops Working, MQTT working fine

soggygeek, 2021-04-22:
From this discussion, two reports of somewhat the same issue (sorry, tons of off-topic postings in the middle of the two reports).  https://groups.google.com/g/electrodacus/c/m0JlWcBBAOk
**Here is the first report:**
I could connect to the SBMS0 via wifi as usual (in my setup I have the SBMS0 connected via my router and the router assigns it a static IP#.)
The home tab was updating as normal, I could also open the settings and the about tabs.
When I tried to look at the legacy view tab, the page was flickering as if it was updating, but showing half or one quarter of the static page, but no data.
I tried with both my laptop and phone, cleared the browser cache on both devices.
**Here is my near-identical experience: (SBMS0 latest hw, Version: 0.3-dirty)**
after about a month or more of operation as normal on the Electrodacus SBMS0 with wifi, with really reliable connection with the wifi, I had the same symptoms this morning.
I am using MQTT from the Electrodacus to a Mosquitto Broker on a laptop to a node-red server showing the data.  I also have another browser tab with the Electrodacus legacy page up.  I mainly monitor the node-red page, because it has a nice graph of what's going on.  But right at charge termination, I like to see the Electrodacus legacy page because it updates faster than my node-red display (which by design only updates every 5-10 seconds).  On the legacy page I can see clearly what's going on with balancing.
This morning, the MQTT publish was WORKING, but the "legacy" web page was not.  I could display all the other pages and they appeared OK (I forgot to notice if the home page data was actually updating).  I tried refreshing the page, I cleared browser cache.  I tried a different browser (on same computer).  Before clearing the cache it was showing some data but not updating, and not all the data (maybe only the SOC).  After clearing cache it was all blank except the basic framework of the page.
We used the "turn wifi on and off" mechanism on the Electrodacus to cycle power on the ESP32 and everything worked after that.
I will report again (and be a little more attentive to details, take screenshots) if it happens again.
SBMS0 
Not a big deal to toggle the wifi board if you are present, but if you are trying to monitor from afar, that's impossible to do.

armageddon421, 2021-04-22:
Hi soggygeek,

This is probably related to issue #23 

I assume that sometimes something goes wrong with the serial communication, maybe due to EMI or similar issues. This causes corrupt data in the buffer of the ESP32 and breaks the legacy webpage. If the issue happens again, please go to /rawdata and paste the contents here so I can have a look.

The history data won't be lost because it is actually stored on the SBMS itself and transmitted completely every second.

ninlee, 2021-06-03:
i have had the sbms0 maybe 6 months or more and i have has the legacy page do this same thing a number of times and turning the wifi on and off again seem to temporarily fix this issue. also after setting up mqtt it worked for 1day and then i cant access the settings to change the mqtt data information and i also cant access the settings under that data menu. see image below. 
how can i fix this. would really like to get the mqtt working again.
[image]

armageddon421, 2021-06-03:
Please go to /rawdata if the legacy page is not working and post the results here.

I am unsure why the settings screen is not working properly. You may try opening the developer console of your browser and look for any script or network errors and report your findings.
You can try updating to the latest version using any of the methods outlined in the readme. This should reset any config and thereby fix any problems with it.

armageddon421, 2021-06-03:
Correction: it may be /rawData with a capital D

ninlee, 2021-06-05:
Problem happened again today. had to reset the wifi to get it back. 

i looked at the developer console in my browser but wasn't able to see any errors given that i am not even sure what to look for.

please see the output of the rawData.

var Btn="zzzzzzz|{xyyy{||}||||||||l]||EE+###%%######################################################################%%%$$####$$$$mh`]XXTTSJEB<843./)()-'/3446047?La]SIFJMXZTLK?APSUZ_ccfnty{}orrnprptvrmspunxz{xx|xy{xxyywwyruuurxvz}z{{yzvtxv{xuuvtvwtqr";
var Btp="################################$####*)27?EFGD=62%#%$$%%'+NJKKHGIMHMJGGGIMPQKIRZ[TNIHIG+#%&%%%&''&%%'+'#########################################################################################################################################";
var ELd="zzzzzzz|{xyyy{||}||||||||l]||EE+###%%######################################################################%%%$$#####$$$mh_]WWSSSHD@:731,.'&',&-2224.25=J`[QGDHLWYSKJ=?PRTY^bcemty{}orrnprptwspvsuoxz{xx|xx{xyyywwxstuurxvz}{{{zywtxv{xvtwtwxtrr";
var Ld ="################################################################################################################################################################################################################################################";
var PV1="################################################################################################################################################################################################################################################";
var PV2="################################################################################################################################################################################################################################################";
var dmppt="############################################################";
var eA="###.?7#################################.<v";
var eW="###%w/#################################%vR";
var gsbms="#########&NN#*k#+F&N7#*X#+0#########";
var s1=['Ah','A','SBMS0  '];
var s2=[0,0,0,0,0,0,0,0,3,6,1,1];
var sbms="8)(4WL#hLqM,M/Lr##LVLfL}*}##-#+4#######*u############$E^%N(";
var xsbms="###QEBr%#$^";

ninlee, 2021-08-13:
i am having problems updating using the methods you outlined in the readme.  i got at far as platformio run and then there is a bunch of errors.

see below

C:\electrodacus-esp32>platformio run
Processing serial (platform: https://github.com/platformio/platform-espressif32.git; board: nodemcu-32s; framework: arduino, espidf)
------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/platformio/platform-espressif32.git
Error: Traceback (most recent call last):
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\commands\run\processor.py", line 70, in process
    p = PlatformFactory.new(self.options["platform"])
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\platform\factory.py", line 54, in new
    raise UnknownPlatform(pkg_or_spec)
platformio.platform.exception.UnknownPlatform: Unknown development platform 'https://github.com/platformio/platform-espressif32.git'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\commands\run\command.py", line 133, in cli
    process_env(
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\commands\run\processor.py", line 72, in process
    self.cmd_ctx.invoke(
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\commands\platform.py", line 325, in platform_install
    pkg = pm.install(
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\package\manager\platform.py", line 51, in install
    pkg = super(PlatformPackageManager, self).install(
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\package\manager\_install.py", line 141, in install_from_url
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\package\vcsclient.py", line 54, in new
    obj = getattr(sys.modules[__name__], "%sClient" % type_.title())(
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\site-packages\platformio\proc.py", line 117, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Lee\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

armageddon421, 2021-08-13:
This is a strange error, as the repository definitely exists.
Are you running the latest version of platformio? Can you try updating everythin?
`pip install -U platformio`
or
`pio upgrade`

ninlee, 2021-08-13:
you tried all that...
You're up-to-date!
PlatformIO 5.1.1 is currently the newest version available.

Everlanders, 2021-08-24:
I just wanted to chime in and +1 this issue. 

We've got an SBMS60 and an SBMS100, these were equipped with a ESP8266 and so I've connected an external ESP32 to the TX pin on the SBMS, and then we're curling the data and storing it in InfluxDB.  (work in progress, any help getting this into InfluxDB appreciated.)

Everything works as you'd expect, except every so often the rawData hangs and we get the last known good values for 1/2 an hour or more. Power cycling everything works, But now that I've seen this I will try and catch the rawData for you.

Everlanders, 2021-08-24:
Well, that didn't take long, The Legacy Page is stuck with the timestamp 22:33 for the last half an hour and the reading aren't updating.

This is the rawData from this event.

var Btn="+**)*-.**))+/+**))-.)*)(*.,)*))*.*))(),.**)(*,-)*))*--*)()*,-*)()0996O}gH;:78555c83000/30/10)'.)))*,&#&#%$##'#-##$##,##([XUF#####%*+####60+&##############&8lg`_V##########NNMI>,###############################################################";
var Btp="#####################################################################################################$#####&###$$#0*##)#####6-.$&###*->C####)8B}omo[JXTJ>'#######%$-17EMMK7######]fXnTjadaXaekt?iktNmjmmq`rlrwmqtpt`ahfah^rjl[qesqspymrzspeqjv|y";
var ELd="&&&&&&&&&&&&&%&&&&&&%&&%&&&&&&&%&&&%&&&&&&%&&%&&&&&&&%&&&%&&%&&&%&&%&?lU8(((((()O(((((((())))(((((()()))())()(()(((((())66568;865678686677667765554:86665434445447769968777778778747:J<FFFIA@@;]>;8U@?:;<I;:::;;:9;CI>FCEE<??M9G9<=::<;:9<H>>888";
var Ld ="('&&'*+&'&&',(''&&)+&'&%&+(&'%&'+'&&%&(+&'&%&)*&'&&')*&&%&')*'&%&-66333548746454:844543:634544;945535:545445/)6)(6+-3+(4def[444<;;<;333;eeeeb]X29;;ANFDCDKKOkd]]W5::::7111=eede_W22;EGEB@BEGEA@CFGDAADGFB@CGDAFDBGAG@GAG@FBDDAFAEDAF@FD@FDACFDAD";
var PV1="######################################################################$%%&'()**+*+-022355224=@>C<><7AQBFCCFL74977GKF57>?;<==<;:9987679=CGJKLNPQPOOOOOOLIEA>:8544579;>@ABBA@@@@ABDGIJOOOOOOOOPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPQQQQQQQQQQQ";
var PV2="#######################################################################$%%&'())*)*,/01134113<>=A;=:6@NADAADJ53766EID46<=9;<<;988765568<AEGHILNNNMMMMMLJGC@<96433457:<>@@@??>>>?@BEGHMMMMMMMMMMMMMMMMMMMMMNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNN";
var dmppt="############################################################";
var eA="##-pYx##-XL7##-PgF########8+5Z##1m>G##)DyV";
var eW="##$J|Y##$IL7##$HMm########%nw&##$|$@###oxi";
var gsbms="#L7#G}#CD$@$#:*#4Q$@u#F[#<|#########";
var s1=['Ah','W','SBMS100'];
var s2=[0,0,0,0,0,0,0,0,8,2,1,1];
var sbms="#%49DQ#TG6G0########G9G:*{##+$G+$F+$?P#Q2###############%N(";
var xsbms="#kAL6>N$#,t";

Everlanders, 2021-08-24:
Moments later it's stuck again... I don't know if another rawData example will help??

`var Btn="+**)*-.**))+/+**))-.)*)(*.,)*))*.*))(),.**)(*,-)*))*--*)()*,-*)()0996O}gH;:78555c83000/30/10)'.)))*,&#&#%$##'#-##$##,##([XUF#####%*+####60+&##############&8lg`_V##########NNMI>,###############################################################";
var Btp="#####################################################################################################$#####&###$$#0*##)#####6-.$&###*->C####)8B}omo[JXTJ>'#######%$-17EMMK7######]fXnTjadaXaekt?iktNmjmmq`rlrwmqtpt`ahfah^rjl[qesqspymrzspeqjv|y";
var ELd="&&&&&&&&&&&&&%&&&&&&%&&%&&&&&&&%&&&%&&&&&&%&&%&&&&&&&%&&&%&&%&&&%&&%&?lU8(((((()O(((((((())))(((((()()))())()(()(((((())66568;865678686677667765554:86665434445447769968777778778747:J<FFFIA@@;]>;8U@?:;<I;:::;;:9;CI>FCEE<??M9G9<=::<;:9<H>>888";
var Ld ="('&&'*+&'&&',(''&&)+&'&%&+(&'%&'+'&&%&(+&'&%&)*&'&&')*&&%&')*'&%&-66333548746454:844543:634544;945535:545445/)6)(6+-3+(4def[444<;;<;333;eeeeb]X29;;ANFDCDKKOkd]]W5::::7111=eede_W22;EGEB@BEGEA@CFGDAADGFB@CGDAFDBGAG@GAG@FBDDAFAEDAF@FD@FDACFDAD";
var PV1="######################################################################$%%&'()**+*+-022355224=@>C<><7AQBFCCFL74977GKF57>?;<==<;:9987679=CGJKLNPQPOOOOOOLIEA>:8544579;>@ABBA@@@@ABDGIJOOOOOOOOPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPQQQQQQQQQQQ";
var PV2="#######################################################################$%%&'())*)*,/01134113<>=A;=:6@NADAADJ53766EID46<=9;<<;988765568<AEGHILNNNMMMMMLJGC@<96433457:<>@@@??>>>?@BEGHMMMMMMMMMMMMMMMMMMMMMNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNN";
var dmppt="############################################################";
var eA="##-pYx##-XL7##-PgF########8+5Z##1m>G##)DyV";
var eW="##$J|Y##$IL7##$HMm########%nw&##$|$@###oxi";
var gsbms="#L7#G}#CD$@$#:*#4Q$@u#F[#<|#########";
var s1=['Ah','W','SBMS100'];
var s2=[0,0,0,0,0,0,0,0,8,2,1,1];
var sbms="#%49DQ#TG6G0########G9G:*{##+$G+$F+$?P#Q2###############%N(";
var xsbms="#kAL6>N$#,t";`

armageddon421, 2021-08-31:
@Everlanders the rawData looks good but must for some reason not be updated anymore. By any chance, do you have MQTT running and can confirm or deny that MQTT is still working in that scenario?

Everlanders, 2021-08-31:
No, I wasn't using proper MQTT, I've been using Blynk (quasi mqtt) (but I'm currently moving away from that as much as I can.)

I was just using this to get the SBMS data onto my network and then I was parsing it in Python to Blynk, InfluxDB, and others.

I have now moved to a wired serial connection.
https://github.com/Burtond/Electrodacus

I was getting several bad readings a day there too, (3485% SoC, etc) but have fought through it, and now have it filtered nicely.

helgef, 2022-08-27:
Just got this as well. Issue present in both the legacy view inside the vue application and also on /sbms.html directly

Seems like the issue us with the javascript parsing the rawData:
Uncaught SyntaxError: Unexpected identifier (at rawData:7:9)

rawData looks like this:

var Btn="###########################################8A-&,8ShA@;WC???7W??7?<XB?9??<U@99Y=<@TH<><77;n}d_`^ZTQenmnmmfgjiigii]YYY^NMRuvvuuuvuutpfbebbbbbbbbbbbbbbbbbbbbbbdno}YZSSRZSUSRW[ZYYZ]Y]W||||||||}{{|{||{||{{{|||{|{{|||||{{||{{{|{|{{|{{||||||{|||{{";
var Btp="%###############$%%%%%%%%&&&&&&'(()((()*,,$#####################################################################################################################################################################################################";
var ELd="###########################################7A,%,8Sg@@;WC???7W??7?;WB>9?><U@89Y<;?TG<><77;n|d_`^YTQemmnmmegjiigii[XXX^MMRuvvuuuvuutofbebbababbbbbabbbbababbbbdno|YZRRRYSURQW[ZYYZ]Y]W{|{||||||{{|{{{{{|{{{|{{z|{{||{||{{||{{{|{{{{|{{||||||{|||{|";
var Ld ="################################################################################################################################################################################################################################################";
var PV1="################################################################################################################################################################################################################################################";
var PV2="################################################################################################################################################################################################################################################";
var dm  t="############################################################";
var dmppt="############################################################";
var eA="##$Os&############################I+##$Im}";
var eW="###4xm############################'z###4/^";
var eWs="##+L6>N$#%}";
var esbms="###L6>N$#%}";
var gsbms="#########$4S$$_#Yn$3w$$(#Y9#########";
var s1=['Ah','A','SBMS0  '];
var s2=[0,0,0,0,0,0,0,0,8,2,1,1];
var sbms="9+>8YD#qF}Fy########FzG#*:##-#Y7#######XY############$C%%N(";
var xsbms="##'L6>N$#%}";

As you can see indeed is line 7 suspect: 
var dm  t="############################################################";

Not sure what happens here.. Might look at the code generating rawData from uart when i get home

helgef, 2022-08-28:
So I traced down what is happening here.

1. The legacy page calls the /rawData inside a <script> tag - So basicly tries to execute whatever is in there as javascript
2. /rawData is simply returning jsvarstore.dumpVars() jsvarstore seems to be some kind of buffer
3. jsvarstore has some sanity checks when inputting new data, but a check for whitespace in var name is missing (Perhaps other illegal varable names also)
4. Since UART messages from the sbms0 does not have CRC or any other form of data integrety checks this mutations are bound to happen sometimes because of noise in transmission
5. Once the corrupt varables are in the jsvarstore they will never be overwritten (as bad data would) because the following messages does not contain the same varable names
6. Thus they are returned every time legacy page is trying to include /rawData causing a javascript runtime error

Proposed fixes:
1. More stringent checks in the varStore which varables are allowed before storing them (check variable names against a list of allowed values?)
2. Clear out the varStore once in while (every minute?)
3. Include CRC or other form of integrety validation to messages (need fixing in sbms0 software)

armageddon421, 2022-08-29:
I'll add the check for spaces and add some kind of timeout to the variables

closed 2022-08-29 by armageddon421

armageddon421, 2022-08-29:
Created a release, the binaries are attached there so you don't have to build yourself.
https://github.com/armageddon421/electrodacus-esp32/releases/tag/0.4

commit 99e62553 (2022-08-29): add jsvar timeout and valid character check

fixed exp32 SDK to 4.4.0

closes #37
closes #23
closes #32