armageddon421/electrodacus-esp32 #34: Unable to compile

glashinc, 2021-07-27:
In following the instructions on how to compile the firmware I run into an issue where the platformio.ini is referring to a branch in the arduino-esp32 repository that no longer exists (https://github.com/espressif/arduino-esp32.git#idf-release/v4.0).  See current branches:

[image]

How do I resolve this?

armageddon421, 2021-07-27:
I think it may just work in the latest master branch. Can you try removing the branch specification?

acgalvan8, 2021-08-05:
I am having the same issue as above.
I changed the platform and platform packages as below:

[env]
platform = espressif32 ; https://github.com/platformio/platform-espressif32.git ;https://github.com/platformio/platform-espressif32.git#feature/idf-v4.0
board = nodemcu-32s
framework = arduino, espidf
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 ; @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0

I am getting this error after it runs:

Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <ArduinoJson> 6.18.3
|-- <PubSubClient> 2.8.0
|-- <ArduinoJson> 6.18.3
|-- <jsvarStore>
|-- <parsers>
|-- <AsyncTCP> 1.1.1
Building in release mode
Compiling .pio/build/serial/src/main.o
src/main.cpp:11:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://platformio.org/lib/search?query=header:Arduino.h
*
*****************************************************************

 #include <Arduino.h>
          ^~~~~~~~~~~
compilation terminated.
Archiving .pio/build/serial/esp-idf/app_update/libapp_update.a
Indexing .pio/build/serial/esp-idf/app_update/libapp_update.a
Archiving .pio/build/serial/esp-idf/asio/libasio.a
Indexing .pio/build/serial/esp-idf/asio/libasio.a
Compiling .pio/build/serial/bootloader_support/src/bootloader_random.o
Compiling .pio/build/serial/bootloader_support/src/bootloader_utility.o
Compiling .pio/build/serial/bootloader_support/src/esp_image_format.o
Compiling .pio/build/serial/bootloader_support/src/flash_encrypt.o
*** [.pio/build/serial/src/main.o] Error 1
Compiling .pio/build/serial/bootloader_support/src/flash_partitions.o
========================= [FAILED] Took 97.16 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
serial         FAILED    00:01:37.162
==================== 1 failed, 0 succeeded in 00:01:37.162 ====================

I am getting the same error on my Macbook and my Lubuntu box.

armageddon421, 2021-08-05:
I can reproduce your issues on a fresh install, it seems my old install had the repos still cached and it just worked for me.
I tried different combinations of repos for 5 hours and just gave up for today. I will look into this and provide a solution soon.

armageddon421, 2021-08-08:
Should be fixed in master branch. Closing this for now, please reopen if issue persists.

closed 2021-08-08 by armageddon421