armageddon421/electrodacus-esp32 #35: DNS resolution MQTT broker

helgef, 2022-05-18:
Hi, new user here.

I was able to get MQTT publishing working yesterday, but only after replacing the hostname of the mqtt broker with its IP. (Im using a cloud based MQTT broker).

Since the IP can potentially change for the broker, I would prefer to use its FQDN. Any hints on how to enable DNS resolution (or should it just work?)

Steps to reproduce:

1. Set mqtt broker to test.mosquitto.com
2. Subscribe with another client to test.mosquitto.com and observe no messages
3. Change mqtt broker address to 35.186.238.101 (DNS resolved IP of test.mosquitto.com)
4. Messages starts appearing in the client

armageddon421, 2022-05-18:
DNS resolution should work out of the box. Maybe there's something wrong with the DNS assignment by your DHCP server?

helgef, 2022-05-18:
Thanks for prompt reply. I will investigate further next time I can access the network. 
I have other devices connected to the same router without DHCP issues though, so seems a little strange.

I am not experiences with the wifi libraries you use, but is there perhaps a possibility to set a static DNS server (e.g 1.1.1.1) for debugging purposes?

armageddon421, 2022-05-18:
I think I did not expose static IP and DNS settings to the user. So without
code changes it is unfortunately currently not possible to set a DNS server
manually.

One thing to try may be flushing the dns records on your router if
possible, sometimes there are issues with that.

helgef, 2022-05-31:
You can close this issue, I have verified that DNS resolution works

closed 2022-05-31 by helgef